engcang/SLAM-application #1: LIO-SAM not working with bag files captured in gazebo

esaumar, 2021-05-31:
Hi @engcang! Nice integration, your videos look pretty good. I have been trying to replay what you did with LIO-SAM repo but using another robot. I am using a differential drive robot model (https://github.com/sanuann/DifferentialDriveRobot) where I attached a Velodyne VLP-16 (https://github.com/lmark1/velodyne_simulator) and a IMU (libgazebo_ros_imu.so). 

I can't get a good reconstruction. Have you seen this behavior? I have found that it could be related to the extrinsic parameters in params.yaml but I believe I'm setting the values correctly. 

[image]

I attached the Velodyne into the robot **chassis** using the file **myrobot.xacro** from https://github.com/sanuann/DifferentialDriveRobot

```
<xacro:include filename="$(find velodyne_description)/urdf/VLP-16.urdf.xacro"/>
  <VLP-16 parent="chassis" name="velodyne" topic="/velodyne_points" hz="10" samples="440">
    <origin xyz="0 0 0.0" rpy="0 0 0" />
  </VLP-16>
```

[image]

I attached the IMU into the **velodyne** body frame and defined an update rate of 500Hz in the file **myrobot.xacro** from https://github.com/sanuann/DifferentialDriveRobot
```
<gazebo>
    <plugin name="imu_plugin" filename="libgazebo_ros_imu.so">
      <alwaysOn>true</alwaysOn>
      <bodyName>velodyne</bodyName>
      <topicName>imu</topicName>
      <serviceName>imu_service</serviceName>
      <gaussianNoise>0.0</gaussianNoise>
      <updateRate>500.0</updateRate>
    </plugin>
  </gazebo>
```

Thus, all sensors are in the same body frame, **velodyne**. That is why I used this Extrinsics

```
# Extrinsics (lidar -> IMU)
  extrinsicTrans:  [0.000, 0.000, 0.0]
  extrinsicRot: [1, 0, 0,
                  0, 1, 0,
                  0, 0, 1]
  extrinsicRPY: [1,  0, 0,
                 0, 1, 0,
                 0, 0, 1]
```
but can't get a good reconstruction. I also tested moving the VLP-16 0.4m above the chassis and modified the Extrinsics to **[0.000, 0.000, -0.438]** (gotten from `rosrun tf tf_echo velodyne chassis`) but I have the same behavior. 
[image]

Any suggestion will be appreciated. Thanks!

esaumar, 2021-05-31:
Btw, I tried to use your script `lidar_repair.py` for the trouble shooting with `CPU ray` in the Velodyne Gazebo plugin. I guess the output is the topic with name `velody` but I don't get any point cloud, not even the red ones I showed in the first comment.

engcang, 2021-06-01:
@esaumar Hi. first thank you for your interest in my repo.
Looking into your questions, I doubt that could because of IMU sensor.
I used `libgazebo_ros_imu_sensor` rather than `libgazebo_ros_imu` and the detailed difference can be found at: http://gazebosim.org/tutorials?tut=ros_gzplugins
According to the original author of LIO-SAM (https://github.com/TixiaoShan/LIO-SAM#prepare-imu-data), you can check that the IMU must include 9-axis values, and I guess it also has to include gravity factor.

Otherwise, exptrinsic parameters seem ok.

esaumar, 2021-06-02:
Hi again @engcang ! Thanks for your quick response. Finally I got consistent reconstructions with LIO-SAM! You were right, it seemed that it was an issue with the IMU. The gravity should be included in the inertial measurements. I changed the plugin from `libgazebo_ros_imu` to `libgazebo_ros_imu_sensor` as suggested and now it works correctly.

[image]
[image]

By using libgazebo_ros_imu_sensor now I can see the gravity on the Z-axis acceleration.
**Bad IMU**

```
orientation: 
  x: 2.27188461314e-06
  y: -4.9678469459e-07
  z: 0.99810920732
  w: -0.0614655208814
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: 0.000769452875622
  y: 0.00324902769181
  z: 4.03222099903e-05
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration: 
  x: 0.227426137416
  y: 0.0242159376913
  z: 4.67897961339
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
```

**Good IMU**

```
orientation: 
  x: 1.55563938577e-14
  y: -6.53325214776e-15
  z: -2.5826076197e-15
  w: 1.0
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: -6.18309291924e-13
  y: 2.5967517029e-13
  z: 6.44077756609e-21
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration: 
  x: 1.2805174418e-13
  y: 3.04905318523e-13
  z: 9.8
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]

```
Thanks for your support!

closed 2021-06-02 by esaumar

narutojxl, 2021-11-11:
Hi @esaumar,  i meet similar problem with [https://github.com/TurtleZhong/LVIO-SAM gazebo](https://github.com/TurtleZhong/LVIO-SAM)
* When use  `libgazebo_ros_imu_sensor.so`, i set its bodyName and frameName all are `imu_link`. When i see with rviz_imu_plugin in rviz, i see  imu's axis is not aligned with `imu_link`, is it normal?  
When i output imu topic, i see its frame id is in `imu_link`, gravity 9.8 is in the `imu_link` 's x axis.
Here is my libgazebo_ros_imu_sensor.so config.
* >	  <gazebo reference="imu_link">
	    <gravity>true</gravity>
	    <sensor name="imu_link" type="imu">
	      <always_on>true</always_on>
	      <update_rate>100</update_rate>
	      <visualize>true</visualize>
	      
	      <plugin name="imu_plugin" filename="libgazebo_ros_imu_sensor.so">
		<topicName>imu/data</topicName>
		<bodyName>imu_link</bodyName>
		<frameName>imu_link</frameName>
		<updateRateHZ>100.0</updateRateHZ>
		<gaussianNoise>0.0</gaussianNoise>
		<xyzOffset>0 0 0</xyzOffset>
		<rpyOffset>0 0 0</rpyOffset>
		<initialOrientationAsReference>false</initialOrientationAsReference>
	      </plugin>
	      <pose>0 0 0 0 0 0</pose>
	    </sensor>
	  </gazebo> 
* Here is `/imu/topic` 
> 	header: 
	  seq: 67514
	  stamp: 
	    secs: 814
	    nsecs:  40000000
	  frame_id: "imu_link"
	orientation: 
	  x: -0.689608468155733
	  y: 0.1563482199301769
	  z: -0.6896015112603228
	  w: -0.15634944975933715
	orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
	angular_velocity: 
	  x: 3.696373038636215e-07
	  y: 3.955931739268561e-06
	  z: 2.899603939727953e-06
	angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
	linear_acceleration: 
	  x: 9.799999999509456
	  y: 3.794165786835077e-05
	  z: -9.026246286142949e-05
	linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
*  infos in top view picture: `imu_link` is x(up), y(right), z(front) with coarse axis. Bule box is rviz_imu_plugin box, imu's axises are the thin and short ones, not aligned with `imu_link`. 
[image]

* vlp-16 config is:
>     
    <xacro:include filename="$(find velodyne_description)/urdf/VLP-16.urdf.xacro" />
    <xacro:macro name="vlp16_mount" params="prefix parent_link topic height:=0.12 *origin">
    	<xacro:VLP-16 parent="${prefix}vlp16_mount_base_link" name="velodyne" topic="/points_raw_in" hz="5" samples="440" min_range="0.1" gpu="false" organize_cloud="false">  
       	<origin xyz="0 0 ${height}" rpy="0 0 0" />
    	</xacro:VLP-16>
    </xacro:macro>
* should lio-sam config `Horizon_SCAN: 1800` or `Horizon_SCAN: 440` ? 
* `extrinsicTrans[]`, `extrinsicRot[]` and `extrinsicRPY[]` are got from `rosrun tf tf_echo velodyne  imu_link` and [online convert tool](https://www.andre-gaschler.com/rotationconverter/)
* > 	 
	rosrun tf tf_echo velodyne imu_link  
	At time 0.000
	- Translation: [0.109, 0.000, -0.764]
	- Rotation: in Quaternion [0.707, 0.000, 0.707, -0.000]
		    in RPY (radian) [-3.142, -1.571, 0.000]
		    in RPY (degree) [-180.000, -90.000, 0.000]

>   extrinsicTrans: [0.109, 0.000, -0.764]
  extrinsicRot: [0, 0, 1,
                 0, -1, 0,
                 1, 0, 0]
  extrinsicRPY: [0, 0, 1,
                 0, -1, 0,
                 1, 0, 0]

Thanks for your help and time sincerely！

narutojxl, 2021-11-11:
The first problem about imu axis not aligned with binded frame axis problem is caused by world file, when using with [cmu indoor.world](https://github.com/HongbiaoZ/autonomous_exploration_development_environment).
The second problem, should lio-sam `Horizon_SCAN` param about vlp-16 be consistent with vlp-16 xacro `samples` param? I see the author's lio-sam `Horizon_SCAN` param still is 1800.

BTW: when use with @engcang's script `lidar_repair.py`, lio-sam's imageProjection node will crash, could you please give a little explaination why should we interchange point cloud's  height - width with ` gpu="false"` in  vlp-16 xacro?  Thanks a lot!