engcang/SLAM-application #4: R3Live drifting

Camilochiang, 2022-07-15:
Great video!
I had try R3Live with similar or worse results, mainly due non structural environments (a.k.a agricultural environments).  
Another option that I see that you have no try in your github is [FAST_LIO_LC](https://github.com/yanliang-wang/FAST_LIO_LC) who implement loop closure. and GPS support  (non mandatory)
I'm thinking to implement it in R3Live but I'm not sure why do the team cast all the data to different point clouds (if you look at your topics there are 100 of color point clouds). Do you know why?

engcang, 2022-07-15:
Thanks for your interest on my tests.

I just skimmed the code of R3LIVE and guess that there is no special reason.
I think that is for just faster visualization since the huge number of pointcloud takes a lot more resources than dividing them into few several topics.

Refer these lines of code: 
https://github.com/hku-mars/r3live/blob/master/r3live/src/r3live_vio.cpp#L986
https://github.com/hku-mars/r3live/blob/master/r3live/src/r3live_vio.cpp#L1031

Camilochiang, 2022-07-15:
Yes that is what I thought. 
Thanks for you input!

closed 2022-07-15 by Camilochiang