engcang/SLAM-application #8: imu coordinate system transformation problem in r3live

jllk47, 2023-11-08:
Hello, I would like to ask how to convert imu coordinate system and lidar coordinate system in r3live. If convenient, can we communicate with each other in other ways

engcang, 2023-11-08:
Hi. As I wrote in here - https://github.com/engcang/SLAM-application#-how-to-properly-set-configuration-for-r3live
LiDAR to IMU calibration is needed.
I think this algorithm is the most recent and good - https://github.com/hku-mars/LiDAR_IMU_Init

jllk47, 2023-11-15:
Hello, I don't see any information about imu calibration here, and there is no configuration of lidar and imu related parameters in your r3live config file. Could you please provide some more detailed information.Thank you