engcang/SLAM-application #10: SLAM selection

aditdoshi333, 2024-02-07:
Hello,  

Thank you for sharing your appreciation for the work. I've followed many of your videos to compare SLAM algorithms and need your help in selecting a SLAM algorithm for my use case.  I have Livox Avia and Livox Mid 360, and we have experimented with the following SLAM algorithms. We need a SLAM algorithm for mapping small and large rooms. However, we are facing issues when generating point clouds for complex spaces that combine large areas, narrow paths, and small rooms.  

- FastLio: Works well in open areas but drifts in narrow paths and small rooms. 
- R3Live: FastLio performs better. 
- Immesh: FastLio is superior. 
- PvLio: Performs better than FastLio, but we are facing issues with Z drift. The whole point cloud bends over the length. It also requires adjusting the voxel size for different sizes of rooms. 

I am considering trying two different approaches: one with a dual LiDAR setup (Avia + Mid and Mid + Mid) and the other with a Mid360 + external IMU. However, I'm not sure which IMU to try.  Could you please share your feedback on which IMU I should consider and what other factors I could explore?  

Thank you very much.

aditdoshi333, 2024-02-07:
The z drift is like this. The bend is more when we reduce the voxel size in PV LIO

[image]

engcang, 2024-02-07:
@aditdoshi333 Hi. I guess you want to map the environments accurately without fast motions, right?
In that case, I think the built-in IMUs in LiDARs are enough. Anyway if you want to get expensive and good one, I recommend XSens IMUs. They have MTi-100, 300, 610, 630, etc. My lab members tried 100, 300, 630 for Visual Inertial Odometry methods and they were all good.

ImMesh uses [VoxelMap](https://github.com/hku-mars/VoxelMap) for its odometry estimate and that is same for PV-LIO.
They estimate probabilistic planes and use them to estimate the odometry. 
So more constraints, should be better performance, more computational resources are used.

But, more constraints do not always provide better performance, as you experienced already.
The matter is how good points are used to estimate odometry depending on the estimation methods they use.

You can refer these papers [RMS](https://github.com/ctu-mrs/RMS) and [Quatro](https://quatro-plusplus.github.io/).
They both point that too many redundant points (or constraints) rather degrade the odometry estimate, as they include more outliers, noises, and worse localizable ones.

As the environments you are targeting include different sizes of spaces (large and small rooms), you can consider to try my lab's Ada-LIO. If you want, send me your datasets. I can test them and if you like the result.
Otherwise, you can develop your own algorithm based on Fast-LIO (I just recommend this algorithm for its compact structure and good performance with fast computations) that uses the good points only depending on the environments where the robots are located.

aditdoshi333, 2024-02-07:
Thanks a lot for sharing your thoughts. 

Please share your email address I would like to share our dataset with you.

engcang, 2024-02-07:
@aditdoshi333 Hi. My email address is: eungchang_mason@kaist.ac.kr

aditdoshi333, 2024-02-09:
Shared over mail