engcang/SLAM-application #11: ikd-tree problem

chenxiaocongAI, 2024-04-16:
hello engcang! Have you solved ikd-tree to only keep the point cloud closest to the center in each grid. This phenomenon will be inconsistent with the actual point cloud, resulting in slight jitter in the pose.

engcang, 2024-04-16:
Hi @chenxiaocongAI 

AFAIK, ikd-tree already keeps the point closest to the center in each grid at here - https://github.com/hku-mars/ikd-Tree/blob/e2e3f4e9d3b95a9e66b1ba83dc98d4a05ed8a3c4/ikd_Tree.cpp#L502-L512, when using downsampling (grid means the downsampled voxel here, right?).
(and when not using downsampling, all points are inserted and maintained at here - https://github.com/hku-mars/ikd-Tree/blob/e2e3f4e9d3b95a9e66b1ba83dc98d4a05ed8a3c4/ikd_Tree.cpp#L549-L570)

If you look through the FAST-LIO2, it already separate the points to be downsampled and not, to efficiently keep the point cloud closest to the center in each grid at here - https://github.com/hku-mars/FAST_LIO/blob/main/src/laserMapping.cpp#L428-L466

engcang, 2024-05-11:
@chenxiaocongAI If you do not have further questions, let me close this issue!

closed 2024-05-11 by engcang