engcang/SLAM-application #13: fast-lio-sam每次建图得到轨迹有差异

seven12767, 2025-01-20:
同一组数据，同一组参数，跑两次，得到的轨迹无法完全重合，用evo_ape计算出来误差3-6cm，请问大佬，算法中哪一部分存在随机性吗？为什么每次运行的结果不一样。

seven12767, 2025-01-22:
> [@seven12767](https://github.com/seven12767) Hey! I am happy to answer your question and want to solve the issue as far as I can understand what your question is. I am not a Chinese nor cannot speak Chinese! It would be great if you can ask in English or Korean :)

I got a ".bag" data，running twice with nothing changed, the two trajectories cannot completely overlap,  there are 3-6cm calcalating with evo_ape tool，I  hope to know why? or which part of the algorithm (FAST-LIO-SAM) has randomness? Thankyou so much for answering my question.

engcang, 2025-01-22:
@seven12767 

Hi! This repository tries to estimate and optimized the pose estimation and trajectory with two main part: LIO and loop-closing.
1. LIO - It adopts Fast-LIO2 and it estimates the odometry with point-to-plane residual iterative error state Kalman filter.
2. Loop-closing - It uses GTSAM to optimize the keyframes trajectory with the transformation calculated with ICP of PCL library.

I guess the latter one, loop-closing is the reason. The implementation of my repositories are trying to detect the loop and calculate the matches between candidate loop in real-time with multi-threading scheme, so that the detected potential loop keyframes pair can differ for each running. Then, the transformation result can differ and the resulted loop-closed trajectories can differ.

In summary, the reason is the real-time loop-closing with multi-threading. 
So if you want to test it, you can modify the code into offline method. I guess then the results of each running will be same.

closed 2025-01-22 by engcang